olofk/fusesoc #530: Thread-safe generators

fantognazza, 2021-11-18:
Fusesoc is not thread-safe when processing generators: the same `.core` file is shared among different executions, and gets overwritten by independent fusesoc runs.

It is a real-case scenario with consistent reproducibility when spawning multiple fusesoc executions with GNU parallel command, even if independent `build-root` parameters are provided.

Here are reported two errors printed out by independent execution runs:

> INFO: Generating ${VENDOR}:${LIBRARY}:${NAME}-memory_generator:${VERSION}
> WARNING: The first line of the core file ${HOME}/.cache/fusesoc/generated/${VENDOR}_${LIBRARY}_${NAME}-memory_generator_${VERSION}/${NAME}-memory_generator.core must be "CAPI=1" or "CAPI=2:".  The first line of this core file is "".
> ERROR: Setup failed : Unknown file type

> INFO: Generating ${VENDOR}:${LIBRARY}:${NAME}-memory_generator:${VERSION}
> ERROR: Setup failed : Failed to parse generated core file ${NAME}-memory_generator.core: Missing 'name' parameter

Assuming that the called generate script itself is thread-safe, it would be desirable to achieve concurrent fusesoc runs containing generators.

olofk, 2021-12-22:
This is a problem indeed. I made a change now so that generators are run in unique temp directories instead. Please let me know if 0469da452a68d44689bcb347708c7f4a775d3e41 fixes the issue

fantognazza, 2021-12-27:
Looks good, the only minor issue I can see now is the generator's temporary directory persisting after the fusesoc command ends.

Calling the tool multiple times, like in the case of a design space exploration, leads to a `/tmp` folder full of orphan directories.
In fact, it can be considered somehow similar to a 'memory leak', being `/tmp` mapped to RAM memory.

olofk, 2022-01-11:
That's a fair point. We need to find some point where it's safe to remove them. Probably somewhere in the edalizer.run function