olofk/fusesoc #531: How to use dependencies from non-default target

mballance, 2021-11-19:
I'm working with some .core files that deal with fairly different collections of source that cannot be processed together by the same tool (think Verilog and documentation source). Cores have dependency chains >1 in depth.
My hope was to use different 'targets' to organize the dependent filesets:

Dep_1_1:
```
filesets:
  rtl:
    # ...
  rst:
    # ...

targets:
  hdl:
    filesets:
      - rtl
  doc:
    filesets:
      - rst
```
Dep_1:
```
filesets:
  rtl:
    # ...
    depend:
      - Dep_1_1
  rst:
    # ...
    depend:
      - Dep_1_1

targets:
  hdl:
    filesets:
      - rtl
  doc:
    filesets:
      - rst
```

TopLevel:
```
filesets:
  rtl:
    # ...
    depend:
      - Dep_1
  rst:
    # ...
    depend:
      - Dep_1

targets:
  hdl:
    filesets:
      - rtl
  doc:
    filesets:
      - rst
```

My expectation was that when I processed `Toplevel` and target `doc`, I would receive the .rst files from `Toplevel`, `Dep_1`, and `Dep_1_1`. What I find is that I actually receive just the files from `Toplevel` and `Dep_1`. 

My understanding is that the following logic prevents the target `doc` from being evaluated for Dep_1_1 because we are not processing the dependencies for a top-level core:
https://github.com/olofk/fusesoc/blob/b05eed5b68e819d219ac9a3fc959bf1fe740ff91/fusesoc/capi2/core.py#L631-L644

I have confirmed that if I add a 'default' target to `Dep_1_1`, those files are brought along as dependencies.

Is there a way to cause evaluation of a named (non-default) target for dependencies of non-toplevel cores? Is restricting dependencies to the default target for dependencies of non-toplevel cores a key assumption of FuseSoc? In other words, would something bad happen if this were supported?

Thanks for your time in reading through all of this!
-Matthew

sjalloq, 2021-11-21:
I don't know the reasoning behind it, but yes, by design only the default target is used for dependencies.

You have two options to work around this, use target based flags or just have separate core files for the two targets.

```yaml
targets:
  default:
    filesets:
      - target_doc ? (rst)
      - target_hdl ? (rtl)
```

imphil, 2021-11-21:
The behavior you're seeing is expected, as documented at https://fusesoc.readthedocs.io/en/stable/user/build_system/dependencies.html#what-happens-if-a-dependency-is-specified.

As @sjalloq mentioned, the typical solution is to use separate cores (often the best solution), or flags: ideally not `target_*` flags, which are a bit fragile, but tool-specific flags (e.g. `tool_verilator`). You can find more documentation at https://fusesoc.readthedocs.io/en/latest/user/build_system/flags.html (this documentation page isn't part of the "stable" documentation yet, but it applies equally to the latest stable version).

mballance, 2021-11-21:
@sjalloq, @imphil, thank you both for your response! And thanks, Phillip, for the pointer to the docs. It's good to understand that this is intended behavior. I'll give more thought to how I can accomplish my goals, and close the issue. Thanks again for your quick responses!

closed 2021-11-21 by mballance

olofk, 2021-11-22:
Despite being closed I'll just add some more meat to this as it pops up from time to time. There are two vaguely related reasons
1. While it's possible to do like you did when you're in control of the whole dependency chain, I believe it would be impossible to come up with a naming convention that everyone could stick to and you would need to inspect each core to find out what names they have used for their filesets.
2. Other people have suggested adding a syntax to depend on e.g. `[core.some_fileset, another_core.different_fileset]`. Unfortunately this will make things much more complicated than what it might first look like, because these two targets can have vastly different (and perhaps conflicting) dependency chains. So in practice we would need to look at each fileset as its separate core to have proper handling for this

As has been mentioned earlier, use flags is the way to go. Unfortunately we have so far only added rudimentary support for it. The functionality works but isn't as ergonomic as we would like. Today, you can run fusesoc e.g. with `fusesoc run --flags=doc --target=doc core` and then add conditionals to your dependencies to pull in different filesets in the default target depending on whether the doc flag is set or not

The next thing to happen is that we will make it possible to set default values of flags in a target. This will allow you to create a target with the `doc` flag set instead of specifying it manually on the command-line. Eventually I'm hoping to add support for local use flags as well so that each core can require its dependencies to have certain flags set or unset. With that you should be able to do something like `depend:["dep_1[doc]"]`. This however requires an improved dependency engine which is something I so far haven't been able to find someone to help with :(

mballance, 2021-11-23:
Appreciate the extra information, @olofk! It sounds like the ability to set flag values in a target is the easier task, and I think this would be a significant step forward. Is the work to support this currently underway, or is this also awaiting a contributor? 
I believe I can make progress using file_type as a filter, but like the concept of using flags as a way to pull filesets with some flags-identified characteristic.

sjalloq, 2021-11-23:
@olofk, how much work is that dependency engine change?  I'd like to help out but still don't know my way around the internals that well.  If it was a side project that I could complete over time that would be okay for me.  I regularly want more flag support and I think target specific auto flags would solve a lot of the pain I see.

olofk, 2021-11-26:
@mballance We're actually pretty much there already. I pushed a PR last April (https://github.com/olofk/fusesoc/pull/502) but never got around to finish it up the changes mentioned there. If you or @sjalloq would find this useful and could take it for a test drive I would find it easier to finish this up. https://github.com/olofk/fusesoc/pull/500 is somewhat related since intended use case is to find invalid use flag combinations.

@sjalloq The current dependency manager is taken from https://github.com/enthought/sat-solver/ and I just massaged it enough to work for FuseSoC. It does its job but is far from ideal. For example, I haven't really figured out how to get any useful error messages which means that all dependency errors look like crap in FuseSoC. And it's definitely not flexible enough to handle local use flags. The whole idea of use flags is taken from Gentoo Linux which has a package manager called portage written in Python. It's possible there might be something to reuse from there but I don't know. There's also an alternative dependency manager called [pkgcore](https://github.com/pkgcore/pkgcore) that might be a good base too. Or maybe something else or write it from scratch. I don't know really. Looking at the integration in FuseSoC however, there's not many functions that need to be implemented. You would need to implement an alternative to the [CoreDB](https://github.com/olofk/fusesoc/blob/master/fusesoc/coremanager.py#L31) class that just have the functions `add`, `find` and `solve`. My gut feeling is that a new depdencency manager can be implemented to a large degree outside of FuseSoC and then just plugged in