olofk/fusesoc #534: Generate coverag information with ghdl

danselmi, 2021-12-17:
Hi,

I like to generate coverage information for a core with ghdl (gcc backend).
For this i need to add these options:
`-Wc,-fprofile-arcs -Wc,-ftest-coverage -Wl,-lgcov`during make phase (ghdl -m) but not for the import (ghdl -i).
GHDL complains if these options are present during import.

When I add these options to targets:sim:tools:ghdl:analyze_options they are added to both phases.

Is it possible to add options only to make phase? Or is there another solution?

m-kru, 2022-04-03:
Do you have an error message for `ghdl -i` when `-lgcov` is added?

danselmi, 2022-04-07:
Yes, I got the following error message: `ghdl:error: unknown command option '-lgcov'`

olofk, 2022-05-31:
The most straight-forward way would be to add a new backend option to split up the `-i` and `-m` options. But!
I was halfway through a rewrite of the ghdl backend some months ago and I have a vague memory that I planned to drop the `-i` `-m` flow in favor of some other flow. Unfortunately I can't remember the details and I never manage to learn how ghdl works. I mention this because there might be bigger changes needed and if so, I want to get this right from the start