olofk/fusesoc #535: what parts of semantic versioning are really supported ?

krishnan-gopal, 2021-12-20:
Hi

If I use a caret or tilde dependency on an external core, fusesoc seems to only consider versions of that core with a pure 'major.minor.patch' format while resolving dependencies.
Versions with a pre-release or build metadata are not considered as semantic in such a case.
Is this intentional ?

imphil, 2021-12-20:
That's correct: the versions are expected as major.minor.patch, as documented at https://fusesoc.readthedocs.io/en/latest/user/build_system/dependencies.html. 

That doesn't necessarily need to stay that way: do you want to attempt a pull request?

krishnan-gopal, 2021-12-21:
Already on it :)

olofk, 2021-12-22:
FYI, I like semantic versioning, but I believe there are a whole bunch of cores that don't use semantic versioning. FuseSoC has tried to encourage semantic versioning but not enforcing it. Also perhaps good to know is that we support a Gentoo-inspired extension, namely a -rx suffix (where x is a number) to indicate a revision. This is normally used when the core description file is updated without the core itself having been changed