olofk/fusesoc #539: Adding tcl cmd prior to implementation run

aignacio, 2022-01-13:
Hey everyone,

thanks for the project, since I started using it, I have been evangelising people about it =). I have a simple question which regards of adding the tcl command below prior to the synthesis/implementation step. I have tried to add in the top of the .xdc but had no luck with Vivado, for every run I see the DRC violations during impl. of unconstrained pins which stops the progress of my build. Can you point me if there's an easy  way of adding it into my core file?

```tcl
set_property BITSTREAM.General.UnconstrainedPins {Allow} [current_design]
```

sjalloq, 2022-01-17:
Have a look at the hooks example in Open Titan: https://github.com/lowRISC/opentitan/blob/master/hw/top_earlgrey/chip_earlgrey_nexysvideo.core

aignacio, 2022-01-19:
Thanks, solved!

closed 2022-01-19 by aignacio