olofk/fusesoc #541: No module named 'edalize.tools.spi_nor_config_generator'

soronpo, 2022-01-20:
Followed the getting started instructions but get the following error when running `fusesoc`
```
Traceback (most recent call last):
  File "/home/tealc/.local/bin/fusesoc", line 5, in <module>
    from fusesoc.main import main
  File "/home/tealc/.local/lib/python3.8/site-packages/fusesoc/main.py", line 28, in <module>
    from fusesoc.coremanager import CoreManager, DependencyError
  File "/home/tealc/.local/lib/python3.8/site-packages/fusesoc/coremanager.py", line 16, in <module>
    from fusesoc.core import Core
  File "/home/tealc/.local/lib/python3.8/site-packages/fusesoc/core.py", line 9, in <module>
    from fusesoc.capi2.core import Core as Capi2Core
  File "/home/tealc/.local/lib/python3.8/site-packages/fusesoc/capi2/core.py", line 908, in <module>
    for backend in get_edatools():
  File "/home/tealc/Downloads/edalize/edalize/__init__.py", line 38, in get_edatools
    return [get_edatool(pkg) for pkg in walk_tool_packages()]
  File "/home/tealc/Downloads/edalize/edalize/__init__.py", line 38, in <listcomp>
    return [get_edatool(pkg) for pkg in walk_tool_packages()]
  File "/home/tealc/Downloads/edalize/edalize/__init__.py", line 22, in get_edatool
    return getattr(import_module("{}.{}".format(__name__, name)), name.capitalize())
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'edalize.tools.spi_nor_config_generator'
```

olofk, 2022-01-20:
I'm not sure what's going on here but I have a theory at least. Edalize finds its backends dynamically by looking at which modules that exist in the edalize root directory. I see that it's looking for a `edalize.tools.spi_nor_config_generator` but no such thing exists in Edalize. Do you know what that is and where it is located? Knowing that would provide another puzzle piece

soronpo, 2022-01-20:
At first I have not installed `edalize` at all, and just followed the `fusesoc` instructions. I also tried installing `edalize` with not much change. I have a VBox machine with Ubuntu 20.04 and docker with 18.04. Both with the same issue.

GuzTech, 2022-01-21:
I have the exact same issue. I installed fusesoc on my work PC `pip install fusesoc` which installed version 1.12.0. It also installed edalize version 0.3.2. Python version is 3.10.

It doesn't matter where I run fusesoc and even what the command line parameters are (even without), it always gives that error. Now on my home PC with the same OS and packages it works just fine.

I cannot find the string `spi_nor_config_generator` in neither fusesoc nor edalize so I have no idea why it's trying to find it.

olofk, 2022-01-21:
Yes, if we can find where that `spi_nor_config_generator` is coming from I hope we can get another clue. Clearly something is weird here. @soronpo can you search your disk for that file? I can't reproduce the issue here so I really need both your help to figure out what's going on

olofk, 2022-01-21:
ok, I did a global search on github and it turns out there's a `spi_nor_config_generator`in the litespi Litex package. The interesting thing is that this module lives in a subpackage called `tools`I wonder Edalize is picking up that for some strange reason

olofk, 2022-01-21:
Yes. Installing this triggers that bug. God, I hate Python sometimes

olofk, 2022-01-21:
I just released an Edalize 0.3.3 that I hope will fix this problem. Happy for any feedback

olofk, 2022-01-21:
FYI, it might be that we're hitting this Python bug https://bugs.python.org/issue14787 that turns 10 years old in less than a month :)

soronpo, 2022-01-21:
Yes, it works now! Thank you @olofk

closed 2022-01-21 by soronpo

GuzTech, 2022-01-21:
I'm at home so I cannot test if it works, but I assume it will since @soronpo said it does :)

soronpo, 2022-01-21:
I explicitly installed edalize, FWIW.

GuzTech, 2022-01-24:
I can confirm too that this problem has been solved. Thank you!