olofk/fusesoc #543: vhdl generics parameters not working with Xsim on windows

Steinegger, 2022-01-28:
I've encountered and issue with the fusesoc vhdl parameters/generics on windows when using xsim. I added a paramters section to my .core file and also referenced this parameter in the simulation target:
```
parameters:
  MY_PARAM:
    datatype: int
    default: 60
    description: "something"
    paramtype: generic
```

If I run the fusesoc command this will result in
`ERROR: [XSIM 43-3225] Cannot find design unit work.60 in library work located at xsim.dir/work.`

The culprit seems to be the way the parameters are generated in the resulting makefile:
```
GEN_PARAMS = --generic_top MY_PARAM=60
```

If I change this to 
```
GEN_PARAMS = --generic_top "MY_PARAM=60"
```
and call the makefile manually with `make run-gui` or `make run` it works as intended.

olofk, 2022-01-28:
Thanks for reporting. Proper quoting of arguments in different simulators is one of those things that are surprisingly hard to get right. I guess it should be a matter of changing [this line](https://github.com/olofk/edalize/blob/master/edalize/xsim.py#L131) to something like `'--generic_top "{}={}"'.format(k, self._param_value_str(v))`

Do you want to give that a try and send a PR (you can use a local Edalize by running `git clone https://github.com/olofk/edalize && cd edalize && pip3 install --user -e .`)? I can do it otherwise but not sure I can test it properly here

Steinegger, 2022-01-28:
Indeed that does the trick on Windows. Thanks for quick reply! I create a PR with the fix you suggested.

closed 2022-02-09 by olofk