olofk/fusesoc #559: Sending an argument value to modelsim backend does not work as expected

hackathlete, 2022-04-01:
Hello,

There is a generic value in my VHLD design that I want to set by sending an argument value to the backend each time I run the simulation (each time another value).

In the .core file the parameter is defined as follows:
```
parameters:
  G_SEED:
    datatype    : int
    paramtype   : generic
```

Then I run fusesoc with the command:
`fusesoc run --run --target=sim --tool=ghdl fusesoc:argument_test:seed_generic:0.0 --G_SEED=$RANDOM`
`fusesoc run --run --target=sim --tool=modelsim fusesoc:argument_test:seed_generic:0.0 --G_SEED=$RANDOM`

The expected behaviour is that each time I simulate the design another random value is passed to the simulator.

With ghdl this works as expected.

However, with modelsim the argument value is not sent to the simulator each time but the first time only. I had a quick look at the makefile for modelsim and it seems, that the argument value from the first invocation is written into the makefile. Instead the argument value from each invocation should be taken.

Attached the files of a short example: 
[seed_generic.zip](https://github.com/olofk/fusesoc/files/8396478/seed_generic.zip)

olofk, 2022-04-01:
Thanks for reporting. I think I fixed it with https://github.com/olofk/edalize/tree/modelsim_runtime_args Would be great if you could give that a try

olofk, 2022-12-18:
@hackathlete Did you get a chance to test the above fix?

olofk, 2023-01-19:
Please reopen if the problem remains

closed 2023-01-19 by olofk