olofk/fusesoc #567: Section "env" in scripts

pantikov, 2022-05-20:
Hello.

I have a question about section "env". It have type Dict of [String] and I write core as example below

```
scripts :
  memory :
    cmd : [/bin/sh, firmware.sh]
    env : { FIRMOBJ: "main" }
```

But package build fails.

```
Traceback (most recent call last):                                                                                                                                   
  File "/usr/local/bin/fusesoc", line 11, in <module>                                                                                                                
    sys.exit(main())                                                                                                                                                 
  File "/usr/local/lib/python3.6/site-packages/fusesoc/main.py", line 776, in main                                                                                   
    args.func(cm, args)                                                                                                                                              
  File "/usr/local/lib/python3.6/site-packages/fusesoc/main.py", line 374, in run                                                                                    
    args.build_root, 
...
  File "/home/user/.local/lib/python3.6/site-packages/yaml/representer.py", line 231, in represent_undefined
    raise RepresenterError("cannot represent an object", data)
yaml.representer.RepresenterError: ('cannot represent an object', 'main')
```
Have I correctly described the "env" section? 
best regards

olofk, 2022-05-31:
It looks like you hit a bug. I haven't analyzed what goes on, but there is something wrong with FuseSoC here. As a workaround for now, perhaps you could set the environment variables before launching FuseSoC, e.g. `FIRMOBJ=main fusesoc run ...`

pantikov, 2022-06-02:
Thanks for answer. Now I use pre-run script for define env.