olofk/fusesoc #570: [Feature Request] Tool version check

armleo, 2022-06-11:
I thinks the FuseSoC should have a feature that checks the underlying tool's version.

For example, I need to use at least Yosys 0.10, but Ubuntu Repos contain Yosys 0.9, as a result, a lot of people install Yosys and get a compilation errors. It would be a nice feature to be able to specify the Yosys version required in the .core file.

imphil, 2022-06-15:
Thanks for this feature request. 

While there is certainly value in having a "generic" and core-specific way to describe tool dependencies, there are some rather interesting challenges with it (e.g. how to get tool versions reliably, how to standardize on version comparision, etc.)

In the meantime, you can create your own solution as a pre_build hook. Have a look at this example:

* https://github.com/lowRISC/opentitan/blob/master/check_tool_requirements.core
* https://github.com/lowRISC/opentitan/blob/master/util/check_tool_requirements.py
* https://github.com/lowRISC/opentitan/blob/d8ba072b3f239cb505c07cb058b97120d26d473f/hw/lint/common.core#L22-L24

mithro, 2022-06-15:
@imphil - The very fact it is hard to get tool versions easily makes it a perfect thing to be part of edalize!

armleo, 2022-06-15:
> In the meantime, you can create your own solution as a pre_build hook
As of right now its done by Makefile. I had to use Makefiles, because I wanted to run multiple targets in the parallel. Should I make a feature request for multiple targets running in parallel?

olofk, 2022-11-26:
It would make sense to have in Edalize, but at the same time, it's not trivial to spec out how it will be used. The computer that runs FuseSoC and the one that launches the EDA tools can very well be two different machines, and this is an intended use-case, so we need to defer the detection until it's actually supposed to run. And at that time, Edalize is not really used anymore, other than running a makefile.

We could look into adding it, but in the short term I agree that @imphil's way of adding it as a pre_build script is the way to go. I use similar scripts myself to first check if the license server is available and things like that