olofk/fusesoc #573: Add support for ternary operator for use with flags

shareefj, 2022-06-16:
The current flag implementation allows the user to optionally include dependencies (or files or set parameters etc) based on the flag being true of false:

```yaml
targets:
  lint:
    <<: *default
    filesets_append:
      - flag ? (optional_fileset)
```

However, if you want to switch between two options based on a flag, then it quickly becomes quite ugly:

```yaml
targets:
  my_target:
    <<: *default
    filesets_append:
      - flag ? (optional_fileset)
      - "!flag ? (alternate_fileset)"
```
 
and this quickly gets out of order when you use more than one flag:

```yaml
targets:
  lint:
    <<: *default
    filesets_append:
      - "!flag_0 ? ( !flag_1 ? ( !flag_2 ? (optional_fileset)))"
```

I think a umber of the issues we've had trying to express a logical combination of flags would be simplified by having a ternary operator:

```yaml
targets:
  lint:
    <<: *default
    filesets_append:
      - flag ? (tru_fileset) : (false_fileset)
```

Is this something worth perusing?

olofk, 2022-11-26:
I want to give this whole expression parsing thing a proper think-through. The current implementation is a subset of the syntax available for Gentoo ebuilds. Originally I was hoping to find a reasonably stand-alone  ebuild expression parser to get all those features for free, but in my investigations it looks like the ones that exist are too deeply embedded into their code bases to be easily extractable.

So failing that, I'm considering some other expression parser that can handle variables, comparisons, conditionals and basic math operations but not much more. Have been trying to find something existing to avoid having to invent a whole new syntax but haven't found anything.

So, right now I'm instead considering to actually spec out a custom syntax. What you suggest could very well be part of that syntax, but let's try to make a simple spec first to avoid extending the current syntax in a way that would make other things harder to implement.

jdwk, 2023-04-26:
If there is a voting system, I would upvote this feature request.

nicorum, 2024-08-27:
> So failing that, I'm considering some other expression parser that can handle variables, comparisons, conditionals and basic math operations but not much more. Have been trying to find something existing to avoid having to invent a whole new syntax but haven't found anything.

@olofk With the pyparsing package already in use, you can relatively easily create your own parser for expressions with the required functionality. There are two interesting examples in the git: [simpleBool.py](https://github.com/pyparsing/pyparsing/blob/master/examples/simpleBool.py), [eval_arith.py](https://github.com/pyparsing/pyparsing/blob/master/examples/eval_arith.py)

Since I am very interested in the possibility of using logical operations, I would also like to help with the implementation. But I agree that there has to be a syntax first. My idea would be to have a conditional expression in infix notation. And put this expression always in parenthesis to not interfere with the flag/!flag syntax. So that such things would be possible:

```
targets:
  lint:
    <<: *default
    filesets_append:
      - ((flagA and flagB) or flagC == (1+2)) ? (true_fileset) : (false_fileset)
```

I can currently see no downside in adding a ternary operator.