olofk/fusesoc #580: Bug: Fusesoc gives error when unrelated core file from llvm is in the directory

hossein1387, 2022-08-01:
I am trying to add Fusesoc to [Ara](https://github.com/pulp-platform/ara). One of the dependency of this project is risc-v LLVM which is populated when the project is cloned. However, calling any Fusesoc command that reads a core file will fail if issued in the top ara directory. 

#### How to reproduce the bug (warning: it takes a long time to clone risc-v gcc and llvm repositories):

      git clone https://github.com/pulp-platform/ara.git
      git submodule update --init --recursive
      git submodule sync --recursive
      touch ara.core
      fusesoc library add ara .
      fusesoc core list

The last command produces this error:

```
Traceback (most recent call last):
  File "/export/tmp/hemmat/anaconda3/bin/fusesoc", line 8, in <module>
    sys.exit(main())
  File "/export/tmp/hemmat/anaconda3/lib/python3.9/site-packages/fusesoc/main.py", line 774, in main
    cm = init_coremanager(config, args.cores_root)
  File "/export/tmp/hemmat/anaconda3/lib/python3.9/site-packages/fusesoc/main.py", line 521, in init_coremanager
    cm.add_library(library)
  File "/export/tmp/hemmat/anaconda3/lib/python3.9/site-packages/fusesoc/coremanager.py", line 233, in add_library
    self._load_cores(library)
  File "/export/tmp/hemmat/anaconda3/lib/python3.9/site-packages/fusesoc/coremanager.py", line 220, in _load_cores
    found_cores = self.find_cores(library)
  File "/export/tmp/hemmat/anaconda3/lib/python3.9/site-packages/fusesoc/coremanager.py", line 209, in find_cores
    core = Core(core_file, self.config.cache_root)
  File "/export/tmp/hemmat/anaconda3/lib/python3.9/site-packages/fusesoc/core.py", line 17, in __new__
    l = f.readline().split()
  File "/export/tmp/hemmat/anaconda3/lib/python3.9/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa8 in position 72: invalid start byte
```     

I was not sure what is happening here. So I just printed the files that are being accessed by Fusesoc in `/export/tmp/hemmat/anaconda3/lib/python3.9/site-packages/fusesoc/core.py` line 17. And sure enough, this file was making Fusesoc unhappy: `riscv-llvm/lldb/test/Shell/Register/Core/Inputs/x86-64-netbsd.core ` which is a binary file. Although I found a way not to include riscv-llvm repository in top ara directory, but I think this issue is beyond this and Fusesoc should handle not related `core` files.

imphil, 2022-08-02:
Thanks for your bug report.

FuseSoC assumes that all `*.core` files in its search path are FuseSoC files. I'd strongly recommend keeping the FuseSoC search path limited to a path that contains RTL files, and not just a full, large directory structure -- if nothing else, it slows down the build process significantly.

FuseSoC already today ignores files it cannot read, but as you've found out, some error modes are not yet caught. Do you feel comfortable opening a pull request that captures the error and ignores the core file?

olofk, 2022-08-02:
If you put a file called `FUSESOC_IGNORE` in a directory, then FuseSoC will ignore that directory and all directories below. So in your case you could probably just do a `touch riscv-llvm/FUSESOC_IGNORE`. I also just pulled in #566 which allows you to specify ignore dirs in your fusesoc.conf instead if you don't want to pollute your llvm with a FUSESOC_IGNORE file.

hossein1387, 2022-08-02:
@olofk that is cool to know! I just created a PR that seems to fix the issue as well.

olofk, 2022-08-03:
This should hopefully be fixed now with #581. Thanks for the report and the fix

closed 2022-08-03 by olofk