olofk/fusesoc #584: Generating a filelist

skilambi, 2022-08-08:
Hi all. I am very new to fusesoc. I was wondering if there is a way to generate a filelist for all the registered cores. Maybe this is a very obvious command. 

Sai.

shareefj, 2022-08-08:
This was discussed in https://github.com/olofk/edalize/issues/25 but it has gone stale.  We're currently using the Filelist backend implemented here: https://github.com/idex-biometrics/edalize/blob/write_filelist/edalize/filelist.py

skilambi, 2022-08-08:
> This was discussed in olofk/edalize#25

Thanks for such a quick response. How do I use this python code? Do you
have an example?

Regards,
Sai.

shareefj, 2022-08-08:
You'll need to copy/fork that repo and pip install it instead of the official Edalize.  I'd suggest having a read of how Fusesoc integrates with Edalize and looking at some of the flows in more detail if you are new to it all.

skilambi, 2022-08-08:
Thanks alot for the quick response Shareef :). Will do.

Sai.

shareefj, 2023-01-18:
@skilambi The Edalize branch was merged to main by Olof so feel free to check it out.  I've been working on a separate branch so may not have merged all the fixes so I'll try to find time to check that.  But if you have any requests or issues then open an Edalize ticket and mention me.

olofk, 2023-01-19:
Yes. I think we can close this and open an Edalize ticket if needed

closed 2023-01-19 by olofk