olofk/fusesoc #589: Question: Many small dependencies from the same repository

flaviens, 2022-10-09:
Hi there!

My apologies if this question was already answered.

## The problem

I'm writing a core file for a project that has many small dependencies [always from the same repository](https://github.com/adithyasunil26/basejump_stl_cores). I'm using fusesoc-12.0.
If I naively provide a list of them, then the same repository is being cloned over and over again, which is very inefficient.
One way around it seems to be creating a big core for the whole repository, but it's not so elegant and I imagine that it's not ideal.

What is the right way only clone the repository once, in this context?

## Steps to reproduce

```bash
mkdir temp
cd temp
touch empty.v
fusesoc library add manydeps .
fusesoc library add basejump_cores https://github.com/adithyasunil26/basejump_stl_cores
cat << EOF >> manydeps.core
CAPI=2:

name: ::manydeps:0.1
description: Many dependencies from the same repo

filesets:
  rtl:
    files:
      - empty.v: {is_include_file : true}
    file_type: systemVerilogSource
    depend:
      - bsg_async_fifo
      - bsg_launch_sync_sync
      - bsg_async_ptr_gray
      - bsg_cache
      - bsg_cache_miss
      - bsg_cache_decode
      - bsg_cache_sbuf
      - bsg_cache_sbuf_queue
      - bsg_channel_tunnel
      - bsg_channel_tunnel_in
      - bsg_channel_tunnel_out
      - bsg_1_to_n_tagged_fifo
      - bsg_1_to_n_tagged
      - bsg_channel_tunnel
      - bsg_channel_tunnel_in
      - bsg_channel_tunnel_out
      - bsg_1_to_n_tagged_fifo
      - bsg_1_to_n_tagged
      - bsg_fifo_1r1w_large
      - bsg_fifo_1r1w_pseudo_large
      - bsg_fifo_1r1w_small
      - bsg_fifo_1r1w_small_unhardened
      - bsg_fifo_1rw_large
      - bsg_fifo_tracker
      - bsg_flow_counter
      - bsg_one_fifo
      - bsg_parallel_in_serial_out
      - bsg_parallel_in_serial_out_dynamic
      - bsg_round_robin_1_to_n
      - bsg_round_robin_2_to_2
      - bsg_round_robin_n_to_1
      - bsg_serial_in_parallel_out
      - bsg_serial_in_parallel_out_dynamic
      - bsg_serial_in_parallel_out_full
      - bsg_shift_reg
      - bsg_two_fifo
      - bsg_cam_1r1w_replacement
      - bsg_cam_1r1w_sync
      - bsg_cam_1r1w_tag_array
      - bsg_mem_1r1w
      - bsg_mem_1r1w_one_hot
      - bsg_mem_1r1w_synth
      - bsg_mem_1r1w_sync
      - bsg_mem_1r1w_sync_synth
      - bsg_mem_1rw_sync
      - bsg_mem_1rw_sync_mask_write_bit
      - bsg_mem_1rw_sync_mask_write_bit_synth
      - bsg_mem_1rw_sync_mask_write_byte
      - bsg_mem_1rw_sync_mask_write_byte_synth
      - bsg_mem_1rw_sync_synth
      - bsg_mem_2r1w_sync
      - bsg_mem_2r1w_sync_synth
      - bsg_mem_3r1w_sync
      - bsg_mem_3r1w_sync_synth
      - bsg_adder_cin
      - bsg_adder_one_hot
      - bsg_adder_ripple_carry
      - bsg_arb_fixed
      - bsg_array_concentrate_static
      - bsg_buf
      - bsg_buf_ctrl
      - bsg_circular_ptr
      - bsg_concentrate_static
      - bsg_counting_leading_zeros
      - bsg_counter_clear_up
      - bsg_counter_clear_up_one_hot
      - bsg_counter_set_down
      - bsg_counter_set_en
      - bsg_counter_up_down
      - bsg_counter_up_down_variable
      - bsg_crossbar_o_by_i
      - bsg_cycle_counter
      - bsg_decode
      - bsg_decode_with_v
      - bsg_dff
      - bsg_dff_en_bypass
      - bsg_dff_reset_en_bypass
      - bsg_dff_chain
      - bsg_dff_en
      - bsg_dff_reset
      - bsg_dff_reset_en
      - bsg_dff_reset_set_clear
      - bsg_dlatch
      - bsg_edge_detect
      - bsg_encode_one_hot
      - bsg_expand_bitmask
      - bsg_hash_bank
      - bsg_hash_bank_reverse
      - bsg_idiv_iterative
      - bsg_idiv_iterative_controller
      - bsg_lfsr
      - bsg_locking_arb_fixed
      - bsg_lru_pseudo_tree_backup
      - bsg_lru_pseudo_tree_decode
      - bsg_lru_pseudo_tree_encode
      - bsg_mux
      - bsg_mux_bitwise
      - bsg_mux_butterfly
      - bsg_mux_one_hot
      - bsg_mux_segmented
      - bsg_muxi2_gatestack
      - bsg_nor2
      - bsg_nor3
      - bsg_nand
      - bsg_priority_encode
      - bsg_priority_encode_one_hot_out
      - bsg_reduce
      - bsg_reduce_segmented
      - bsg_rotate_left
      - bsg_rotate_right
      - bsg_round_robin_arb
      - bsg_scan
      - bsg_strobe
      - bsg_swap
      - bsg_thermometer_count
      - bsg_transpose
      - bsg_unconcentrate_static
      - bsg_xnor
      - bsg_mesh_stitch
      - bsg_noc_repeater_node
      - bsg_wormhole_concentrator
      - bsg_wormhole_concentrator_in
      - bsg_wormhole_concentrator_out
      - bsg_wormhole_router
      - bsg_wormhole_router_decoder_dor
      - bsg_wormhole_router_input_control
      - bsg_wormhole_router_output_control

targets:

  default:
    filesets: [rtl]

  lint:
    default_tool: verilator
    filesets: [rtl]
    tools:
      verilator: 
        mode: lint-only
EOF

fusesoc run --target=lint manydeps
```

Thank you!
Flavien

olofk, 2022-11-25:
One way, as you noted, is to treat the repository as a library where all cores are local cores (i.e. without provider sections in the .core files). This has the drawback that you can't easily use one version of some core and another version of another core since the library will be monolithic. Another potential option is to use the url provider instead of the github provider, where you can point to an url of the file you want to fetch, Currently, the url provider only support downloading a single file, but that should be extended anyway to allow getting several files. This also made me wonder if it's possible to download a subdir of a git repo as a zip/tarball and apparently this has been thought about before, although there isn't any official solution as far as I can see https://stackoverflow.com/questions/7106012/download-a-single-folder-or-directory-from-a-github-repo (someone mentions checking out the repo as an svn repo, which might actually be a reasonable way to do it)

flaviens, 2022-11-28:
Hi Olof, thank you for your response! It helps a lot.

closed 2022-11-28 by flaviens