olofk/fusesoc #592: Question: How to cache VUnit files between builds?

creiter64, 2022-11-16:
Hi all,

I have fusesoc setup to build my core, including a target to run testbenches via VUnit.

My *.core file looks something like this:

```
CAPI=2:
name: fancy::testcore:0.0.1
description: Fancy test core

filesets:
  rtl:
    files:
      - design/fance_core.vhd
    file_type: vhdlSource
    logical_name: abc

  tb:
    files:
      - testbench/fancy_core_tb.vhd
    file_type: vhdlSource
    

targets:
  default: &default
    filesets:
      - rtl
    toplevel: fancy_top

  # The "sim" target simulates the design.
  sim: &sim
    # Copy all key/value pairs from the "default" target.
    <<: *default
    description: Simulate the design
    default_tool: vunit
    filesets_append:
      - tb
    toplevel: fancy_top_tb
    tools:
      vunit:
```

Now the problem is, that if I execute the `sim` target, the complete VUnit dependency is always re-compiled, which takes significantly longer than running the testbench itself. Is there any way to chache the compiled VUnit library?
This behavior occurs with both GHDL and Questa, so I think this has something to do with FuseSoc.

dustex, 2022-11-21:
I have the same issue and would be very interested if there is a way around this.

olofk, 2022-11-25:
Yes, this is a weakness with FuseSoC. It is something I hope to fix next year though. Working on a roadmap so it's easier for everyone to see