olofk/fusesoc #595: How to cache generator output

shareefj, 2022-12-02:
I've been trying to use generators for the first time and am struggling.  One of the main issues is a recent change in #530 that forced the use of transient temporary directories.

In my case I'm trying to compile some external C++ library code using CMake in order to feed that into Verilator.  What I really don't want to happen is that the generator is run for every invocation.  I want the library generated once and reused across subsequent calls to fusesoc.

I can see how to do this on the generator side by using `self.files_root` as my output directory and then checking for the existence of my output file before rebuilding, but how should I handle the core file?  I could add absolute paths to the generated core file but I know that's not recommended and support might be being deprecated?

Any thoughts on the best approach here?

olofk, 2022-12-17:
I am to work on this after the 2.0.0 release

olofk, 2023-02-23:
I am happy to announce that this is now implemented and documented. Please open a separate bug if there's anything in particular that doesn't work as expected

closed 2023-02-23 by olofk