olofk/fusesoc #596: Question: How to define provider for dependencies in core file

dustex, 2022-12-06:
Hello,

I'm looking for the best way how to specify the provider of a core without using the "library add" fusesoc command.

**Outline**
- My project is split into several cores, where each core has its own git repository, let's cal them submodules.
- Some of the submodules have dependencies on other submodules. 
- All these submodules have VUnit-based test benches, and their simulation is run via fusesoc in the CI.
- In addition, there is a git repository with the top-level project, which has dependencies on these submodules.

What I'm doing at the moment to resolve the dependencies of a submodule is to execute `fusesoc library add some-lib git@gitlab.foo.bar:project/vhdl-utility-library.git` in the `gitlab-ci.yml` before I run the fusesoc simulation target. Now, my problem is that I have to place the same `library add` command in the `gitlab-ci.yml` for every submodule which depends on this library and of course also in the top-level project. This is a classic example of code duplication with all its downsides, which I want to avoid (if for instance the gitlab URL of a submodule changes, I have to update every single `gitlab-ci.yml` file where the corresponding submodule is used).

Is there a way around this issue?

Best regards,
Christian

olofk, 2022-12-17:
There are some long-term improvements to be made here, but I'm afraid none of that will happen very soon.

One solution you could perhaps use in the meantime is to have a single library (in your toplevel repo, or in a completely separate repo) with remote cores (like e.g. the standard library https://github.com/fusesoc/fusesoc-cores). You will however run into another type of code duplication this way if you still want to have .core files in each of the repos.

dustex, 2022-12-23:
Olof, thanks for the information!

closed 2022-12-23 by dustex