olofk/fusesoc #601: fusesoc gets stucked while searching recursively in a specific path

aignacio, 2023-01-07:
Hey all,

I have an issue with my current setup where I add some files from [this](https://github.com/alexforencich/verilog-ethernet) repository and when I'm running the target (for instance to build through vivado) `fusesoc` gets stucked forever. I've tried to remove the `example` folder where it contains a symlink to an internal path ([like here for instance](https://github.com/alexforencich/verilog-ethernet/tree/master/example/ATLYS/fpga/lib)) and this solved the problem. However as I include this repository as a submodule, I'd have to remove such path every time I want to use fusesoc when cloning the repository. I suspect that fusesoc is stucked in some kind of loophole with the symbolic link.... thus, my question is: **Is there a way to ignore a path through the .core file?**

olofk, 2023-01-16:
It's a known issue and was very recently fixed in version 2.0rc1. pip won't install that version by default, but if you run `pip install --user -U --pre fusesoc` it should get installed

aignacio, 2023-01-17:
Thanks @olofk

olofk, 2023-01-19:
Released version 2.0 now that should have fixed this. Please reopen if the problem remains

closed 2023-01-19 by olofk