olofk/fusesoc #605: `--build-root` argument modifies more than just the root name

shareefj, 2023-02-13:
I don't know if this is a bug or was by design, but it doesn't seem intuitive and actually adds work for the user.

The `fusesoc build` sub-command has a `--build-root` argument that allows the user to modify the root of the build directory for a particular invokation.  However, this modified build directory doesn't follow the same hierarchy naming convention which means that script that want to build multiple versions of an executable also have to modify the path they search for generated files.

If I call `fusesoc build VLNV`, the resulting build directory is named:

```build/V_L_N_V/target_tool```

but if I do the same with the `--build-root my_build` I get:

```my_build/target_tool```.

olofk, 2023-02-23:
I'm not sure myself if it should be considered a bug or a feature. It is definitely asymmetrical but it was done for a reason. My thinking was that if you as a user set the build-root, you want the path to be predictable, as you perhaps run fusesoc from within a script or so that needs to do something with the output dir. By removing the V_L_N_V component, it makes it much easier to predict that, as you only need to append `target_tool` to get the work root (or just `target` if you use the new flow API). In the regular case (without setting build-root) however we want to have the V_L_N_V component so that you can build several cores in one workspace without having them overwrite each other.

Perhaps this long answer should go directly into the docs instead :)

olofk, 2025-08-21:
We have `--work-root` since FuseSoC 2.3 which should be more predictable and I recommend using that instead.