olofk/fusesoc #609: Exception raise by FuseSoC 2.0 for valid YAML

shareefj, 2023-03-03:
There seems to be a couple of matching comments in the Gitter channel but no solutions and I can't find any matching tickets open.  So here is one but feel free to close if it's a duplicate.

We've just updated to the latest release and am finding that at least one flow now falls over.  I'm running Verilator with the following options:

```yaml
  sim:
    <<: *default
    default_tool: verilator
    tools:
      verilator:
        mode: cc
        verilator_options:
          - '-CFLAGS "-DTOPLEVEL_NAME=chip_sim_tb"'
          - '-F common.args'
          - '-F trace.args'
          - '-F comms.args'
```

And FuseSoC spits out the following exception:

```     
Traceback (most recent call last):
  File "/home/shareefj/git/genesee_asic_merging/.venv/bin/fusesoc", line 8, in <module>
    sys.exit(main())
  File "/home/shareefj/git/genesee_asic_merging/.venv/lib/python3.8/site-packages/fusesoc/main.py", line 740, in main
    fusesoc(args)
  File "/home/shareefj/git/genesee_asic_merging/.venv/lib/python3.8/site-packages/fusesoc/main.py", line 730, in fusesoc
    args.func(cm, args)
  File "/home/shareefj/git/genesee_asic_merging/.venv/lib/python3.8/site-packages/fusesoc/main.py", line 293, in run
    run_backend(
  File "/home/shareefj/git/genesee_asic_merging/.venv/lib/python3.8/site-packages/fusesoc/main.py", line 412, in run_backend
    edam = edalizer.run()
  File "/home/shareefj/git/genesee_asic_merging/.venv/lib/python3.8/site-packages/fusesoc/edalizer.py", line 104, in run
    self.create_edam()
  File "/home/shareefj/git/genesee_asic_merging/.venv/lib/python3.8/site-packages/fusesoc/edalizer.py", line 194, in create_edam
    self.flags["tool"]: core.get_tool_options(_flags)
  File "/home/shareefj/git/genesee_asic_merging/.venv/lib/python3.8/site-packages/fusesoc/capi2/core.py", line 404, in get_tool_options
    options[member] = self._parse_list(
  File "/home/shareefj/git/genesee_asic_merging/.venv/lib/python3.8/site-packages/fusesoc/capi2/core.py", line 741, in _parse_list
    _x = x.parse(flags)
  File "/home/shareefj/git/genesee_asic_merging/.venv/lib/python3.8/site-packages/fusesoc/capi2/core.py", line 67, in parse
    self.exprs = Exprs(str(self))
  File "/home/shareefj/git/genesee_asic_merging/.venv/lib/python3.8/site-packages/fusesoc/capi2/exprs.py", line 154, in __init__
    self.ast = _parse(string)
  File "/home/shareefj/git/genesee_asic_merging/.venv/lib/python3.8/site-packages/fusesoc/capi2/exprs.py", line 143, in _parse
    raise ValueError(
ValueError: Invalid syntax for string: Expected end of text, found '"'  (at char 8), (line:1, col:9). Parsed text was '-CFLAGS "-DTOPLEVEL_NAME=chip_sim_tb"'.
```

shareefj, 2023-03-06:
Seems like the `exprs.py:_get_parser()` needs to be updated to cope with double quotes.  See #610

olofk, 2023-03-06:
Yes. #610 applied now. Can we close this?

cd-vishal, 2023-03-13:
Hey @olofk , I read about this project from FOSSI GSOC'23 idea list and would love to work on JS package of this Library. Can you please tell me some other learning resources other than this repo to learn more about the project? thanks in advance.

closed 2024-06-11 by shareefj