olofk/fusesoc #613: Windows non-C:-Drive issues with .core files

Steinegger, 2023-03-24:
I encountered a ValueError exceptions if the fusesoc project is located on a drive that isn't the C: drive and a generator is called that creates a .core file, or a used library is located on another drive letter. The latter can be easily worked around, but I couldn't find a way to overwrite the `cache_root` directory to get generators to work. Is there a way to set it to a path relative to the directory, like the build folder? The issue seems to be that there's no relative path between the two locations.

Detailed exception:
```
Traceback (most recent call last):
  File "C:\Users\username\AppData\Local\Programs\Python\Python310\Scripts\fusesoc-script.py", line 33, in <module>
    sys.exit(load_entry_point('fusesoc==2.0', 'console_scripts', 'fusesoc')())
  File "C:\Users\username\AppData\Local\Programs\Python\Python310\lib\site-packages\fusesoc\main.py", line 721, in main
    fusesoc(args)
  File "C:\Users\username\AppData\Local\Programs\Python\Python310\lib\site-packages\fusesoc\main.py", line 711, in fusesoc
    args.func(cm, args)
  File "C:\Users\username\AppData\Local\Programs\Python\Python310\lib\site-packages\fusesoc\main.py", line 286, in run
    run_backend(
  File "C:\Users\username\AppData\Local\Programs\Python\Python310\lib\site-packages\fusesoc\main.py", line 405, in run_backend
    edam = edalizer.run()
  File "C:\Users\username\AppData\Local\Programs\Python\Python310\lib\site-packages\fusesoc\edalizer.py", line 103, in run
    self.create_edam()
  File "C:\Users\username\AppData\Local\Programs\Python\Python310\lib\site-packages\fusesoc\edalizer.py", line 176, in create_edam
    rel_root = os.path.relpath(files_root, self.work_root)
  File "C:\Users\username\AppData\Local\Programs\Python\Python310\lib\ntpath.py", line 718, in relpath
    raise ValueError("path is on mount %r, start on mount %r" % (
ValueError: path is on mount 'C:', start on mount 'D:'
```

Steps to reproduce for generators:
- Put a fusesoc project on the D: drive (this can be done using the subst command if no other drive is available)
- Call target that has a generator and yields a .core file. 

Steps to reproduce for libraries:
- Put a fusesoc project on the D: drive (can be done with subst if no other drive letter is available)
- Put a library with a .core file that is used in the first project to another Drive letter
- Add the library with `fusesoc library add`
- run a target in the project on the D: drive

Versions:
Python 3.10.0
Fusesoc 2.0
edalize 0.4.1

olofk, 2023-04-10:
Aha! This is not something we have thought about. Thanks for reporting. I think we need to add an option to control where the cache ends up.

FriedrichKiesel, 2025-07-25:
I found two ways to avoid this issue:
1.	Set the environment variable XDG_CACHE_HOME to the directory you want the cache to be generated in. Other tools might search for this variable as well and other data than the cache may end up in your folder.

2.	Add a section like the following to your fusesoc.conf:
```
[main]
cache_root = D:/tmp
```
Also you need to give your configuration to fusesoc via the –config option, otherwise the setting has no effect