olofk/fusesoc #616: Generator-Cache removal and --no-export

Steinegger, 2023-03-27:
Files generated by a generator are removed after the setup phase. When using the --no-export flag the source files aren't copied to the respective build directory, hence, the project creation fails due to missing source files.
I also tried with, the `cache_type: input` and `cache_type: generator` setting to keep the fusesoc config in place [as mentioned in the docs here](https://fusesoc.readthedocs.io/en/latest/user/build_system/generators.html), but they would still be deleted.

This seems unintended behaviour, because deleting the files if the--no-export flag is present will almost always brick the build.
A not so obvious workaround seems to be to specify a copyto attribute for each file in the generated core, if this is the intended way of handling generators it should be noted in the docs.

Versions:
Python 3.10.0
Fusesoc 2.0
edalize 0.4.1

olofk, 2023-04-10:
Confirmed. The --no-export case slipped through our net when we were testing. Sorry about that. I have a potential fix [here](https://github.com/olofk/fusesoc/tree/no-export-generators-fix) and would love some additional testing to see that I got it right this time.

You can install a local development version of FuseSoC by cloning and checking out the branch with
`git clone https://github.com/olofk/fusesoc -b no-export-generators-fix`, cd into fusesoc and run `pip install --user -e .`

Steinegger, 2023-04-12:
Thanks for the quick fix. I removed my copyto statements and installed the dev-version. The fix creates a `generator_cache` directory in the `build/*/*/` folder and works perfectly with the no-export flag. Without --no-export the files are created int the `C:\users\<username>\.cache\fusesoc\generator_cache\` directory and copied over to the build directory afterwards.

Interestingly it also helps with issue #613, because the generated directory is moved to the build directory

olofk, 2023-12-15:
Looks like we can close this

closed 2023-12-15 by olofk