olofk/fusesoc #617: Cached compiles use stale source files

McSherry, 2023-03-29:
This issue seems to be new in v2.1 as I can't reproduce it when downgrading to v2.0.

I have a core that runs a set of tests using Verilator and Googletest. In v2.1, if I change one of the test files, the old version is still used. For example, I have a core file:

```yaml
CAPI=2:

name: oxionics:ip:ieee802p3u-4b5b:0
description: IEEE 802.3u 4b/5b encoders

filesets:
  rtl:
    files:
      - rtl/ieee802p3u_4b5b_tx.sv
      - rtl/ieee802p3u_4b5b_rx.sv
    file_type: systemVerilogSource
  
  tb:
    files:
      - tb/test.cc : {file_type: cppSource}
      - tb/tx.cc : {file_type: cppSource}

targets:
  default: &default
    filesets:
      - rtl
  
  tb_tx:
    <<: *default
    toplevel: ieee802p3u_4b5b_tx
    default_tool: verilator
    filesets_append: [tb]
    tools:
      verilator:
        mode: cc
        exe: "true"
        verilator_options:
          - -Wall
          - -CFLAGS -std=c++20 -I/usr/src/gtest/include
          - -LDFLAGS -lgtest
```

And the minimum needed to reproduce test file:

```c++
#include <gtest/gtest.h>

TEST(tx, example) {
    FAIL();
}
```

If I run `fusesoc run --target=tb_tx oxionics:ip:ieee802p3u-4b5b` then, as expected, it builds and Googletest reports a failure.

If I comment out `FAIL();` and rerun that command, Googletest _still_ reports a failure. The test only succeeds if I `rm --rf build/` and then rerun the `fusesoc` command. This would seem to indicate that it's using the cached, stale version of the test source file.

If I add `--no-export` to the `fusesoc` command, then it fails to build because it looks under `build/` rather than where the source file  actually is.

davidthings, 2023-04-07:
I got a new machine, reinstalled the latest Fusesoc and am experiencing the same thing.  Downgrading to 2.0 works.  Thanks McSherry.

olofk, 2023-04-10:
I think I found the problem now, at least under the assumption that it affects Verilator specifically.

The build is a two-stage process. First, the RTL files are "verilated" into C++ files and a <project>.mk file is created. Then, the verilated C++ files together with the testbench are compiled into an executable.

The Makefile generated by Edalize had a dependency on the testbench files for the first step, but the testbench files are really only used in the second step, so there was nothing to pick up the changes.

This is now (hopefully) fixed with https://github.com/olofk/edalize/commit/19736c9dc9cc68c38cef266c562bb1a371dd220c
and even provides a bonus benefit. If only the testbench is modified, the verilator process doesn't need to rerun which saves a bit of time for larger models.

Would be great if you could test by updating to the latest Edalize development version and report back. One thing that does confuse me a bit is that it looks like you are using the old tool API instead of the newer flow API and I thought this bug only affects the flow API (which is where I also fixed the issue) so maybe we have several bugs lurking here.

McSherry, 2023-04-11:
I cleaned my virtualenv and then installed FuseSoc v2.1 to confirm I could still reproduce the issue:

```sh
$ pip freeze | xargs pip uninstall -y
$ pip install -r requirements.txt
$ pip install --upgrade fusesoc==2.1
```

I could reproduce it, so I installed a new version of edalize and cleaned my build directory:

```sh
$ pip install --upgrade git+https://github.com/olofk/edalize.git@19736c9
$ rm -rf build/
```

Unfortunately, using the tool API (`fusesoc run --target=... ...`) this seems to still be broken.

It's not really clear to me how to rearrange this for the flow API, but if you can suggest a replacement `.core` file section I'll happily test it out.

Barabas5532, 2023-04-12:
I can reproduce this issue using icarus, it's not specific to verilator. The rebuild happens as expected using 2.0.

olofk, 2023-04-13:
Thanks @McSherry and @Barabas5532  for confirming. I will dig into this.

@McSherry If you want to try with the flow API, to provide another data point, you can change your target to

```
  tb_tx:
    <<: *default
    toplevel: ieee802p3u_4b5b_tx
    flow: sim
    filesets_append: [tb]
    flow_options:
      tool: verilator:
      verilator_options:
        - -Wall
        - -CFLAGS -std=c++20 -I/usr/src/gtest/include
        - -LDFLAGS -lgtest
```
(Technically it's fine to keep the tool and tools keys also. The flow key takes precedence if both are present)

olofk, 2023-04-13:
To begin with, I can confirm the behavior at least :/

olofk, 2023-04-13:
Ok, I think I got a fix now. @davidthings @McSherry @Barabas5532, could you please try this branch? https://github.com/olofk/fusesoc/tree/rebuild_fix

Barabas5532, 2023-04-15:
That branch is working for me, thanks @olofk.

closed 2023-04-17 by olofk