olofk/fusesoc #621: Python 3.6 SyntaxError in edalize

shenki, 2023-04-12:


shenki, 2023-04-12:
This shows up in fusesoc ci: https://github.com/olofk/fusesoc/actions/runs/4634094461/jobs/8199945878

buehlerIBM, 2023-04-12:
I am the colleague. I am 100% new to fusesoc and don't know what shenki's comment should tell me.
I installed python3.9 (and 3.8) and made it the default version. However, fusesoc is hard-coded to use python3.6. If I try to overwrite that e.g. with `python3.8 ~/.local/bin/fusesoc` it still won't run:
```
python3.8 ~/.local/bin/fusesoc
Traceback (most recent call last):
  File "/home/buehler/.local/bin/fusesoc", line 7, in <module>
    from fusesoc.main import main
ModuleNotFoundError: No module named 'fusesoc'
```
I guess this is because of this: `~/.local/lib/python3.6/site-packages`
I played around with symbolic links etc. but couldn't get it to work. I guess I'll have to wait for someone to help me.
BTW: I am running on RedHat8.2

shenki, 2023-04-12:
Try reinstalling fusesoc with the updated python

dbesse, 2023-04-17:
This has been reported in edalize as well, but with no reaction: https://github.com/olofk/edalize/issues/374

buehlems, 2023-04-20:
We could solve the problems on our side: I had changed the python version to 3.9 but not the pip version. Once I fixed that, fusesoc did work.  
So it is basically a documentation problem that remains: Python 3.6 is not sufficient. I verified that 3.8 and 3.9 do work.

dbesse, 2023-04-20:
The problematic file is not used (contains a reference to `edam["top_core"]` which is not defined) and is the only one using python3.8 syntax. So, I believe it would make more sense to fix or remove it instead of requiring 3.8 and calling it a documentation problem.

olofk, 2023-12-14:
I believe this should be fixed with https://github.com/olofk/edalize/issues/374 . Please reopen otherwise

closed 2023-12-14 by olofk