olofk/fusesoc #623: Select stage to run in core file

GCHQDeveloper211, 2023-04-17:
Currently if you wish to only build a core file to a particular stage (setup, build, run) you can only do so through the CLI, using the equivalent arguments.

I propose adding a "stage" element to a build target. Much like the CLI option, this will be optional, and if it is not provided it will default to "run" and like the "tool" option or the "pnr" option, if a CLI argument is provided, that will take priority.

This would end up looking like:
```targets:
    build:
        default_tool: [tool]
        toplevel : [to_level_entity]
        filesets: [fileset_list]
        parameters: [ENABLE_AXI]
        stage: [setup OR build OR run]
        tools:
           .....
```
The reason we are thinking about this, is FuseSoc's concept of "running" a synthesis/pnr tool is to program the FPGA, and that can be quite inconvenient as a default behaviour.

olofk, 2023-04-17:
The behaviour of trying to program the FPGA is indeed a bit problematic and inconsistent between different backends and has been bugging me for a long time (see e.g. here https://github.com/olofk/fusesoc/issues/319 ) . I do have a slightly different idea of how to solve this though. My proposal is to never program by default, and instead add a flow option, say `--pgm` to the relevant Edalize flows to opt-in programming. flow options can be set in the target already so that we can program by default if we want to.

While we're on the topic, I would also like to see some configurability when it comes to the device programming tools so that we can select which programming tool to use and set options

Ideally I would also like to see an external library that handles the FPGA programming instead of having that in Edalize, so that _every_ build tool (Edalize, SiliconCompiler, Litex etc) could share that code.

GCHQDeveloper211, 2023-04-18:
If this were to happen, would that mean that the "build" stage and the "run" stage of a PNR build essentially become the same thing?