olofk/fusesoc #628: fusesoc 2.x on windows: export files get removed due to /, \ mismatch

schwende, 2023-05-01:
After doing a fusesoc run --setup I always find my src/ directory empty with only the folders left, but without any file.

After debugging the issue I found out, that all the files get removed by the "# Clean out leftover files from previous builds" (capi2\core.py line 129). 

Line 133: os.path.relpath(_abs_f, dst_dir) and src_files use other /, \ conventionand therefore the two strings are never matching and all the files get removed.

I'm sure there are better solutions, but I fixed it locally at my place by replacing line 133 with:
if not os.path.relpath(_abs_f, dst_dir).replace("\\","/") in src_files:

hackathlete, 2023-05-13:
Hello,
I faced the same problem with fusesoc 2.1 under Windows 10 and Pyhton 3.9.1. The described change og line 133, solved the issue. Thanks @schwende!

olofk, 2023-11-17:
Ouch. This looks bad. Let's go with the quick fix for now and I will do a new release. Then we can think about a long-term solution. As you mention, it feels a bit hacky and the risk is it will break again at some point :/

olofk, 2023-11-17:
FuseSoC 2.3 is released which should hopefully fix this. Please let me know if it works and thanks everyone for reporting and commenting

olofk, 2023-11-18:
#655 was reported as fixed so I'm assuming this one is fixed as well. Please reopen otherwise.

closed 2023-11-18 by olofk