olofk/fusesoc #631: Can't install fusesoc on Ubuntu 16.04

HelpDesperatelyNeeded, 2023-06-22:
I need to install it for OpTiMSoC. When I run the command it tells me to run: "`pip3 install --user --upgrade fusesoc`". It says this:
```
Collecting fusesoc
  Using cached https://files.pythonhosted.org/packages/fa/0b/f6e27d1ad64ed12b7f1f7bb081f83d3e9d2f704f385f0bea4d492c0f90c8/fusesoc-2.2.1.tar.gz
    Complete output from command python setup.py egg_info:
    Couldn't find index page for 'setuptools_scm' (maybe misspelled?)
    No local packages or download links found for setuptools_scm<7.0; python_version < "3.7"
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-38wsk0ap/fusesoc/setup.py", line 61, in <module>
        python_requires=">=3.6, <4",
      File "/usr/lib/python3.5/distutils/core.py", line 108, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 269, in __init__
        self.fetch_build_eggs(attrs['setup_requires'])
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 313, in fetch_build_eggs
        replace_conflicting=True,
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 826, in resolve
        dist = best[req.key] = env.best_match(req, ws, installer)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1092, in best_match
        return self.obtain(req, installer)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1104, in obtain
        return installer(requirement)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 380, in fetch_build_egg
        return cmd.easy_install(req)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 657, in easy_install
        raise DistutilsError(msg)
    distutils.errors.DistutilsError: Could not find suitable distribution for Requirement.parse('setuptools_scm<7.0; python_version < "3.7"')
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-38wsk0ap/fusesoc/
```
I am using python version:
"Python 2.7.12"
pip version:
"pip 8.1.1 from /usr/lib/python2.7/dist-packages (python 2.7)"
pip3 version:
"pip 8.1.1 from /usr/lib/python3/dist-packages (python 3.5)"

If someone could help or give me an idea of what to do that would be much appreciated.

imphil, 2023-06-22:
Ubuntu 16.04 has been out of support for quite a while now (similar to OpTiMSoC, actually). But either way, OpTiMSoC has a good chance of working with newer versions of Ubuntu. It probably makes sense, however, to use an older version of FuseSoC with it as well.

HelpDesperatelyNeeded, 2023-06-22:
Thanks for the quick response. What version of FuseSoC should I get and how should I install it? I've been trying to get 1.9 to install for a while but it seems to have a lot of dependencies that need to be different versions and I can't quite get it right.

Also, I would prefer not to update Ubuntu. Do I actually need to get a new version or can FuseSoC and OpTiMSoC still work with 16.04?