olofk/fusesoc #638: Does fusesoc support VS SpyGlass？

Wendellpuppet, 2023-08-04:
Hi olofk, I would like to konw if fusesoc supports the tool of VC SpyGlass?

imphil, 2023-08-05:
Yes, see https://fusesoc.readthedocs.io/en/stable/ref/capi2.html#spyglass for an example.

closed 2023-08-05 by imphil

Wendellpuppet, 2023-08-07:
> Yes, see https://fusesoc.readthedocs.io/en/stable/ref/capi2.html#spyglass for an example.

I saw SpyGlass in it, but I didn't VC SpyGlass. I wander if there are both supported with the same way, because I think there are a little difference between these two tools(I think VC SpyGlass is updated version of SpyGlass).

imphil, 2023-08-07:
You'll need to give it a try and potentially do a pull request with an update, I don't think any of the maintainers have access to (VC) Spyglass (any more).