olofk/fusesoc #642: unable to run fuseSOC in windows installation

Aaronyap2002, 2023-08-25:
Hello everyone, I was attempting to install fuseSOC in my windows 11. The download and installation went well until I ran the "fusesoc --version" command. It says that ...

> 'fusesoc' is not recognized as an internal or external command, operable program or batch file.

What can i do to fix this issue?
I have python 3.10.11 and pip version 23.2.1 installed.

[image]

olofk, 2023-08-28:
Do you have it added to your PATH? That's the only thing I can think of

Aaronyap2002, 2023-08-29:
i decided to just give up installing on windows and try in Ubuntu instead

hackathlete, 2023-09-22:
Did you run the command prompt as Administrator?