olofk/fusesoc #653: Add support for vvp extended arguments

Barabas5532, 2023-10-24:
https://man.archlinux.org/man/vvp.1.en#EXTENDED_ARGUMENTS

VVP supports "extended arguments", which can be used to pass runtime information such as plusargs to the simulation. Looks like it is not currently possible to pass these in using fusesoc (https://fusesoc.readthedocs.io/en/stable/ref/capi2.html#icarus).

The extended arguments come after the `inputfile` in the command line, but the `vvp_options` from the fusesoc core file seem to come before the input file.

Is this something that fusesoc should support? I'm open to contribute this feature if it is desired.

olofk, 2023-11-17:
All simulation backends support plusargs, but they are specified in the parameters section instead so that they can be used for any simulator. [Here](https://github.com/olofk/serv/blob/main/servant.core#L544) is an example of how to define a plusarg and how to [enable it in a target](https://github.com/olofk/serv/blob/main/servant.core#L452)

As for other extended arguments, such as -vcd or -lxt, those are support through the `vvp_options` key. But! I just realized now that there is something wrong with the Edalize documentation and it is not getting updated automatically and the online version is very old by now. Will need to fix that.

Barabas5532, 2024-04-27:
`vvp_options` doesn't seem to do what you describe. Adding `-vcd` to it results in the following invocation of vvp: `vvp -n -M. -l icarus.log  -vcd spdif_rx_0.0.1 -fst`. The `-vcd` and other extended arguments must come after the input file name, so it should be something like `vvp -n -M. -l icarus.log  -vcd spdif_rx_0.0.1 -vcd -fst` instead.

In the [source code](https://edalize.readthedocs.io/en/latest/_modules/edalize/icarus.html#), `-vcd` must be added to the EXTRA_OPTIONS part of the template, which currently only contains plusargs. I guess a new yaml field like `vvp_extended_arguments` must be created for this to work. It can be prepended to the plusargs.

olofk, 2024-05-26:
Hmm... in all my years of using Icarus I had completely missed that. As I understand it though, you could still pass these options as plusargs. The only difference between passing `+vcd +fst` (which is how Edalize would invoke vvp if they are specified as plusargs) and passing `-vcd -fst` (which is currently not supported) is that in the latter case these vars are not available through $test$plusargs/$value$plusargs in the design.

Worst case we can add extended options if we need to. Just want to make sure that we can't solve it with the existing options first.