olofk/fusesoc #656: fusesoc export xsa file

isomoye, 2023-12-17:
I usually build petalinux after generating hardware. right now fusesoc only exports the bitstream which means i need to open up the project later on and export the full xsa. Would we be able to just add the xsa export too.

olofk, 2023-12-18:
I think the easiest way is to add some tcl script that does the xsa export. You can add custom tcl files by setting `file_type : tclSource`. Now, the trick is to make sure the tcl script is executed at the right time. I'm not familiar with xsa in particular, but generally speaking there are two options here. Either, it's just a property that you set for the project, and then you will get your xsa file, or it's an extra command that needs to be run at the right point. For the latter, which I suspect is the case here, you would need to define a script hook in your custom TCL script so that Vivado will pick it up at the right time. Let me see if I can find some example of that.

garankonic, 2024-01-16:
In the top-level project, you can create a dedicated target which build xsa (without bitstream):
`filesets:
    xsa:
        files:
          - scripts/export_xsa.tcl: { file_type: tclSource }
targets:
    xsa_target:
        toplevel: your_toplevel
        filesets:
          - xsa
        flow: vivado
        flow_options:
          part: your_part
          source_mgmt_mode: All
          jobs: 1
`
In this way XSA script will be the last one to execute. The script itself looks as follows:
`
# switches to vhdl top-level
set_property target_language VHDL [current_project]
# generate output products
generate_target all [get_files *.bd]
# export xsa
write_hw_platform -fixed -force -file file.xsa
`

isomoye, 2024-01-17:
This solution worked. Thansk @garankonic . closing issue.

closed 2024-01-17 by isomoye