olofk/fusesoc #657: Tool documentation went missing

ttsmit, 2023-12-20:
Eighter I am unable to search the web properly, or the trusted CAPI2 documentation went missing from the Read The Docs. Apart from the [pattern properties](https://fusesoc.readthedocs.io/en/latest/ref/capi2.html#pattern-properties), the extensive list of available tools and their options is gone.

olofk, 2023-12-20:
No, you are right. The main reason is that this information depends on what version of Edalize you are using,so when the FuseSoC docs are generated, we pick up whatever version of Edalize that is available at that point in time. We keep this information in Edalize now, [here](https://edalize.readthedocs.io/en/latest/edam/api.html#tool-options) for the old tool API and [here](https://edalize.readthedocs.io/en/latest/edam/api.html#flow-options) for the new Flow API. I will make sure that there are better pointers to redirect here from the FuseSoC docs

ttsmit, 2023-12-21:
That makes sense. Thank you!

olofk, 2024-01-11:
I added some pointers in the CAPI2 docs now https://fusesoc.readthedocs.io/en/latest/ref/capi2.html Couldn't figure out how to make real links though, but I'm sure someone smarter than me can solve that :)

closed 2024-01-11 by olofk