olofk/fusesoc #662: Is there any way to get the output from a generator or chain generators?

shareefj, 2024-01-04:
Hey,

I've only ever used simple generators before that perform a single job and are used as part of a full Edalize tool flow.  But I have a use case where I'd like to get the generated output, in this case a Verilog file, and either pass it onto my own scripts or run a secondary generator.

The use case is that I'm trying to sanitise some Migen so I'm running a first stage generator that writes out a Verilog netlist (with all the ugliness that this entails) and I'm trying to auto-generate a nice wrapper module that cleans up the interfaces.

Is there any way to run the default target of a core file and grab the path to the generated output or just return the EDAM YAML?  Or is there a way to chain two generators together such that I can pass the output products of the first to the second?

Cheers.

olofk, 2024-01-11:
Unfortunately not in a nice way at least. Other people have asked for the same thing, and we actually sat down and tried to figure out a way to chain generators a couple of years ago, but never got to a good conclusion on how to do that.

I have three suggestions for workarounds, in case any of those are of help.

1. Make a larger generator that does both tasks. If you also want to use them separately, perhaps you can structure the code so that it's possible to share the business logic
2. Call FuseSoC from inside the generator. This might become messy with paths to libraries and such, but at least the newer versions of FuseSoC can more easily be integrated directly into Python projects without having to call the command-line
3. If you are using the new flow API in Edalize, you could perhaps define the generators as frontend tools instead. That way you can chain them together in front of the ordinary flow.

shareefj, 2024-01-11:
Thanks, I'll look at those options.

closed 2024-01-11 by shareefj