olofk/fusesoc #669: Any way to use Fusesoc logging in custom scripts?

shareefj, 2024-01-28:
If I'm writing a script that uses `fusesoc.fusesoc.Fusesoc`, is there any way I can get a logger object to work using the default FuseSoC handler?

```python
    import logging
    from fusesoc.fusesoc import Fusesoc, Config
    from fusesoc.main import _get_core, prepare_work_root

    config = Config()
    fs = Fusesoc(config)
    core = _get_core(fs, "v:l:n:v")

    logger = logging.getLogger()
    print("This prints, obvs")
    logger.info(f"This doesn't, why")

    ... script continues
```

olofk, 2024-03-07:
Good question. I have never really understood how the logging module really works.

Many of the modules have this line `logger = logging.getLogger(__name__)` in it. Does that make any difference?

shareefj, 2024-03-07:
Heh, I was not expecting "I have never really understood how the logging module really works.".  :-)

No, I tried that.  It seems that the logging setup limits logging to only a few specific modules. It does that here: https://github.com/olofk/fusesoc/blob/main/fusesoc/utils.py#L134  Why?  I have no idea and I'm in the same boat as you, I've never fully understood how advanced logging works.  It continues to be a black art that I am not skilled in.

I assume that the person who wrote the logging features is no longer contributing?

olofk, 2024-03-07:
I looked into the logs and it turns out @benreynwar did it with f4fad9131dabefedf830b40ecb3dfc6687131d86 back in 2017. Any clues, Ben? :)

benreynwar, 2024-04-04:
I am also not skilled at logging!  Which is probably why creating the logger didn't work with the code I wrote.  @shareefj I expect the solution is to replace the logging code in utils.py with something better.

benreynwar, 2024-04-05:
I don't remember.  My guess is that I thought the fusesoc logging configuration should be limited to the fusesoc ecosystem.  I don't see a problem with removing that limitation.