olofk/fusesoc #673: Generator Cache doesn't work with --no-export

nicorum, 2024-02-21:
When using the --no-export argument the generator cache gets placed in the `build/<core>/<target>-<tool>/generator_cache `folder instead of the `.cache/fusesoc/generator_cache` folder. But this folder is removed at the start of each fusesoc run. Which is why the generator_cache does not work with the --no-export argument.

I can fix this problem for me by removing `self.gen_root` from the line that contains `self.gen_root or self.core.cache_root,` in the `Ttptttg` class. But I don't want to create a pull request with the possible fix. Because I don't understand what the exact purpose of using self.gen_root is. Is this behaviour intended?

olofk, 2024-03-07:
Hmm... good question. There is some complexity in that part of the code, so I can't honestly say right away if it's a bug or the intended way. One thing though, did you mark your generator as cacheable? That also needs to be done.

nicorum, 2024-03-12:
I'm using `cache_type: input` in the generator core.
Caching works with this generator when not using the --no-export flag.

If you can't reproduce the problem, I can create and upload a minimal example.

PR #704 (2024-12-18): Fixed Generator Cache with --no-export
Fixes #673
The issue was caused by 90500180872ed26c3bee2c41c50be5ff0b49c2b4
I also checked, that this fix does not break  #616 again
closed 2024-12-18 by olofk