olofk/fusesoc #674: Builtin fusesoc parsing cores tree command

newfrogg, 2024-02-26:
Recently, I am dealing with [ibex-demo-system](https://github.com/lowRISC/ibex-demo-system). There are many cores and I find it hard searching for RTL in each .core file because they loop deeper in vendor directory. In addition, name of core file someway are different from the core names. E.g: core name: "lowrisc:ibex:rv_timer" for rv_timer.core. Thanks.

So I wonder:
- Are there any commands we can use to parse the core tree.
- And if existed, can we move through nodes of tree it quickly (like g-f in vim)

olofk, 2024-03-07:
There isn't any built-in command for that, but I agree it would be handy. Especially with IDE integration.

I have a WIP here https://github.com/olofk/fusesoc/blob/filters/fusesoc/filters/dot.py that generates a dot (gv?) file that describes the core tree, so it is definitely possible to add a variant of that, which could also print out all files associated with each core.

hongshui3000, 2024-08-02:
I also have such a need. When looking at a complex design, there are many core files in the project. I need to quickly find the top-level core file (there may be multiple top-level core files) and the dependencies between the core files. This is very helpful for studying complex projects.

johnathan-convertino-afrl, 2024-11-12:
This would be great, my dream would be something like...

`fusesoc core-info --depends spinalhdl:cpu:veronica_axi:1.0.0`

--depends, add dependencies, or just do this by default without it?

OUTPUT
```
CORE INFO
Name:        spinalhdl:cpu:veronica_axi:1.0.0
Description: <No description>
Core root:   ./ip_git/VexRiscv_fusesoc
Core file:   veronica_axi.core

Targets:
default : <No description>

Dependencies:
spinalhdl:repo:spinalhdl:1.10.1 
spinalhdl:repo:vexriscv:1.10.1
AFRL:utility:generators:1.0.0 
```

or maybe core-info --depends --dot to generate the above AND output a gv/dot graph?

I'd like to work out how this should function. I can spend some time on this feature and do a pull request when its done.