olofk/fusesoc #682: Generator instance that depends on other cores

ivanvig, 2024-03-20:
Im using systemrdl to implement my cores CSRs, with a generator that consumes a rdl file and compiles it to systemverilog files and thats working fine.

Now lets say I have 3 cores: A, B and C. The C core rdl file includes A and B rdl files, in order to be able to compile the C core rdl file I need to provide the compiler the paths where it should search for the included files.

My first thought was using the generator's files_root and core_file fields to be able to find the core from which the generator is called and then use some internal fusesoc function to resolve all dependencies from the core and get all the files required. But in order to do that I need to create a new fusesoc instance with a config that I cant know from the generator scope.

So, is there a way to solve this problem where the generator instance requires files from other cores in order to produce source code?

olofk, 2024-03-27:
Generators can't use information from other cores. For tasks that affect several cores I would recommend doing this as an Edalize frontend instead. Perhaps look at the sv2v tool class which receives SV files and produces Verilog files, but in your case it would consume rdl files and create SV files instead.

ivanvig, 2024-03-28:
Great! I will look into it. Once I have it, are you interested in a pull request with that feature in the edalize repo? Otherwise I think I can use it as an external plugin.

olofk, 2024-03-28:
That would be great. I'm likely to need something like that myself when I start some new project. But as you have noted, there is nothing preventing you from keeping it in an external plugin if you so wish. Btw, is this using @amykyta3's excellent PeakRDL tooling?

I also remembered that I did something similar to this but with IP-XACT not too long ago. Let me see if I can dig up an example of that for reference.

amykyta3, 2024-05-05:
I would probably lean more towards option 2.

The main PeakRDL command-line interface is definitely the more "productized" and user-oriented workflow. As I develop the PeakRDL tools, I will be pushing users more towards that interface as the primary and preferred way to interact with the various PeakRDL sub-tools. This also has the benefit of implementing the user/org configuration mechanism described here: https://peakrdl.readthedocs.io/en/latest/configuring.html

Sub-tools will still always have their own documented and stable API for power-users.

amykyta3, 2024-05-05:
> Now lets say I have 3 cores: A, B and C. The C core rdl file includes A and B rdl files, in order to be able to compile the C core rdl file I need to provide the compiler the paths where it should search for the included files.

I would recommend treating RDL as distinct target filesets in the fusesoc file manifests. I usually discourage users from using \`include  directives in the source RDL, and instead use multi-file compilation. But even so, it probably makes sense to provide a mechanism for users to specify include search paths for the RDL preprocessor too.

olofk, 2024-06-10:
Include directives shouldn't be a problem as we can mark the RDL files that need to be included with `is_include_file : true` in FuseSoC. Agree that multi-file compilation is to prefer though.

@jamesrbailey Are you intending to release this plugin as open source? I'm working with peakRDL on a project right now and have been considering making such a plugin myself, but would be better if there is something to build upon instead.

jamesrbailey, 2024-06-10:
It's unlikely I will have anything available to open source in the near or medium term, although that would be my intent.

jamesrbailey, 2025-10-29:
Hopefully what follows is even partially coherent.

I developed a PeakRDL Edalize tool plugin and managed to get something sufficiently functional but it has some gross hacks in it. I think there is probably a better way to do things, and perhaps some more general edalize & fusesoc topics to be considered here. I'll outline the general approach I've taken - hopefully I can share my full source at some point although I would caution that it is probably not a good starting point for the "correct" solution. I also hope to hear from others what they think is the correct approach.

I've written a single Edalize plugin that is encapsulates all PeakRDL subcommands ( - I question whether this was the right approach). The user can then invoke this in fusesoc as a tool frontend. Here is a snippet of a core file for a test simulation that first invokes peakrdl as a frontend.
```
targets:
    default:
      flow: sim
      flow_options:
        tool: verilator
        cocotb_module: sanity
        verilator_options: ["--timing", "trace ? (--trace)","trace ? (--trace-fst)","--trace-params","--trace-structs","--trace-max-array 1024", "--unroll-stmts 80000", "-Wno-WIDTHTRUNC", "-Wno-WIDTHEXPAND", "-Wno-TIMESCALEMOD", "-Wno-MULTIDRIVEN"]
        run_options: ["trace ? (--trace)"]
        frontends: ["peakrdl", "cmake"]
        peakrdl_cpuif: "apb4-flat"
        peakrdl_tops: ["design_top"]
        peakrdl_subcommands: ["ralcpp", "python", "regblock"]
        ...
```

Note that I am invoking three PeakRDL subcommands in this example. In my Edalize plugin I generate an EdaCommand for each of them - there are some smarts in the plugin in terms of passing arguments to PeakRDL based on the subcommand that is being used.

A point of frustration: the python and ralcpp subcommands would initially not run since they only generate python and C++ outputs, which the sim flow here does not think it requires and therefore Edalize (quite smartly) knows not to invoke them. I worked around this by having my PeakRDL Edalize plugin generate dummy System Verilog files in for these subcommands so they don't get dropped. A huge hack that cannot be right - but it is working for me for now.

Going through this I became distinctly aware of some of the limitations that fusesoc and Edalize have when it comes to constructing a build flow that might traditionally be written in something like Make. That said, I also think that there is a hypothetical "standard" pre-edatool build flow (ex: process registers -> compile C/C++ source code -> EDA tool) that could be well supported as an Edalize flow that would satisfy most use cases.