olofk/fusesoc #687: fusesoc is not working for tool=QuestaSim in Linux RedHat.

yvishal-123, 2024-05-20:
Hello all,
I am running a "NOT Gate"(language is verilog) as an example and i defined the tool is QuestaSim. but i am facing the serval issues:-

Traceback (most recent call last):
  File "/home/user2/.local/bin/fusesoc", line 8, in <module>
    sys.exit(main())
  File "/home/user2/.local/lib/python3.6/site-packages/fusesoc/main.py", line 672, in main
    fusesoc(args)
  File "/home/user2/.local/lib/python3.6/site-packages/fusesoc/main.py", line 662, in fusesoc
    args.func(fs, args)
  File "/home/user2/.local/lib/python3.6/site-packages/fusesoc/main.py", line 316, in run
    edam_file, backend = fs.get_backend(core, flags, args.backendargs)
  File "/home/user2/.local/lib/python3.6/site-packages/fusesoc/fusesoc.py", line 139, in get_backend
    backend_class = get_edatool(flags["tool"])
  File "/home/user2/.local/lib/python3.6/site-packages/edalize/edatool.py", line 41, in get_edatool
    return getattr(import_module(f"edalize.{name}"), name.capitalize())
AttributeError: module 'edalize.questasim' has no attribute 'Questasim'

please give me the solution for this issue.

olofk, 2024-05-26:
The name of the backend you are looking for is called `modelsim`, not `Questasim`

yvishal-123, 2024-06-11:
This issue is resolve by some other method and working fine and I really thanks for your support.

closed 2024-06-11 by yvishal-123