olofk/fusesoc #691: Is there any way to pass a *value* to an `Edatool`'s`tool_options` attribute via the command line

Jonah-Foley, 2024-06-10:
I have implemented my own `Edatool`, which has a number of `TOOL_OPTIONS`, i.e.

```
class MyTool(Edatool):

    TOOL_OPTIONS = {
        "file_types": {
            "type": "str",
            "desc": "File types which flist will search for",
            "list": True,
        },
        "target" : {
            "type" : "str",
            "desc" : "The tool which you wish to target (A consume or .f files)",
        },
    }
```

I understand this allows me to write something like the following in my `.core` file:

```
 mytool:
    <<: *default
    flow: generic
    flow_options:
      tool: mytool
      file_types:
        - systemVerilogSource
        - verilogSource
      target: MyTarget
```

These options are then accessible within the `MyTool.setup()` method via `self.tool_options.get("target", "default")`. 

However, I then have a command line interface which goes `argparse -> fusesoc -> invoke my tool`, where i'd like to be able to set the `target` option from the command line. 

I can set fusesoc flags from the command line invocation, so I can do something like:

```
 mytool:
    <<: *default
    flow: generic
    flow_options:
      tool: mytool
      file_types:
        - systemVerilogSource
        - verilogSource
      target: 
        - tool_verilator ? (TargetA)
        - tool_xcelium   ? (TargetB)
        - tool_xyz       ? (TargetC)
```
however then `target` is a list, not a singular value making it prone usage error. This seems like a non ideal solution.

Other solutions seem to not work because after the `edam` object is created all other arguments are thrown away - we therefore cannot pass anything *into* the `EdaTool.setup()` method.

So my question - is there any way to directly set the value of `target` via a command line invocation of the tool?

olofk, 2024-06-10:
Either I'm misunderstanding something, or the answer is very simple. I hope for the latter, but expect the former.

All tool options are available on the command line automatically, so you can do `fusesoc run --target=mytarget core --target=whateveryouwant`. Generally, you can find all tool options available for a target with `fusesoc run --target=mytarget core --help`

Jonah-Foley, 2024-06-11:
I should have clarified, I was not using the `fusesoc` command to run my target directly. I had my own command line interface which took into a `.core` file and internally created a `Fusesoc` object which invoked the `Edatool`. It seems the solution to my problem is to pass args to the underlying `Edatool` via the `backend_args` route. I hadnt come across that till now.

closed 2024-06-12 by Jonah-Foley