olofk/fusesoc #694: Dependencies and target-specific files

suppamax, 2024-06-20:
Let's say I have an IP with some files specific for simulations and others which are specific for synthesis.
They would normally reside in 2 different filesets.
Let's also say that the synthesis tool doesn't like some of the sim files an viceversa.

If this IP is imported in a bigger project. 
My understanding is that the only the file sets defined for the `default` target in the submodule will be imported. 
Which means that I will always end up with one of the tools complaining.

Is there a workaround?

olofk, 2024-06-25:
There are basically two ways around this and I wrote a bit about it in the FuseSoC manual. However, I realized this version of the manual is not in a released version, but you can find it here https://fusesoc.readthedocs.io/en/latest/user/knowledgebase.html#dependency-tree-for-a-core-with-optional-components

Basically, there are two ways. Either split out your sim/synth-specific files into separate cores that depends on the common files and then depend on either the sim- or synth-specific cores from your toplevel. Or use flags, e.g. in your default target of your dependency you write `filesets: [ common_files, "sim_only? (sim_files)", "synth_only? (synth_files)"]` and then in your toplevel targets set `flags: {sim_only: true}` for the sim target and `flags: {synth_only: true}` in your synth target.

Let me know if that works for you.

suppamax, 2024-07-03:
Thanks @olofk !

closed 2024-07-03 by suppamax