olofk/fusesoc #697: What is the file_type for SDF file

desorya, 2024-06-27:
Hi there,

I'm trying to run gate-level simulation with existed netlist on VCS so that I need to involve a standard delay file.
I build a fileset like below but I don't know the exact file type for it.
```
filesets:
  sdf:
    files:
      - single_macro/env/MLM_macro_top_A.func_ssg_Cworst_T_m40.sdf.gz
```
I tried `sdf`, `sdfSource`, `standardDelayFile` and `DELAYFILE` but nothing worked. The sdf file still not shown on generated file list.

Thank you!

olofk, 2024-07-05:
The VCS backend doesn't have support for SDF files yet.

I'm not a VCS user myself, but perhaps I can suggest a workaround. If you set `file_type : user` and also `copyto : .`. This will copy the SDF file to your work directory. If there is also some command-line option for loading SDF files, you can put that in `vcs_options`.

Best would of course be to have SDF directly supported, but someone needs to implement that.

desorya, 2024-07-05:
The function that VCS uses to load the sdf file is `$sdf_annotate()` which only accepts string literal as input. If we use relative path as input, a FileNotFound error will be triggered because the fusesoc collects the source files and copy and put them into a new folder, which will disrupt the relative positions of the files.
In this case, we have to either use absolute path or add sdf file(with `.gz` surfix) into `.core` file and create a proper reference. In many cases, using absolute path is not a good choice because the path will vary from user and needs to be modified frequently. Including the sdf file in the `.core` file and referencing it in the tb should be a better approach, but the question here is how should we achieve that?