olofk/fusesoc #702: simplesat dependency is broken

shareefj, 2024-09-05:
This might be temporary but highlights the issues surrounding not pinning a dependency to a particular version.  The latest update to simplesat is broken and means pip installing fusesoc fails.

https://github.com/enthought/sat-solver/issues/286

andreondra, 2024-09-13:
Simplesat wheel is fixed, I tried installing fusesoc and it works again. However, I also think the version should be pinned to avoid similar problems in the future.

olofk, 2025-02-13:
Things like this can always happen. Pinning the version in FuseSoC is not the right solution as we want the latest version in the most cases. I know other projects (looking at you riscv-config) that pins some dependency to a random version which causes problems in a larger context. I think it has to be up the end users to avoid or work around these kind of issues.

closed 2025-02-13 by olofk