olofk/fusesoc #714: [Feature request] Use a specific branch from a git repository as a fusesoc library

rbrglez, 2024-12-22:
I want an option to specify which branch of `some-lib` repository will be downloaded with `fusesoc`.
Quick and dirty example how could it be implemented:

```bash
fusesoc library add some-lib "-b some-branch git@github.com:some-lib/some-lib.git"
```

rbrglez, 2024-12-22:
Link to pull request: https://github.com/olofk/fusesoc/pull/715

wwqwq2313, 2024-12-22:
Hello, we tried to solve the issue.

This is what we did:

Added a new '--branch' option to the 'library add' command to allow specifying which branch of a repository should be downloaded. This is implemented in the parser_library_add arguments and the add_library function.

You can review changes in this commit: https://github.com/wwqwq2313/olofk-fusesoc-714/commit/74054b5646d877c8da3c6ce35a1965d79f4e09d6.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

obruendl, 2024-12-28:
Ideally it would be possible to not only specify a *branch* but any ref (i.e. a *branch*, *tag* or *commit*).

Especially the option to specify an exact tag would be super helpful because it is common sense to tag releases. Hence, users could easily choose which release they want to add (by selecting the tag).

obruendl, 2024-12-31:
I noticed this option being available in the latest version.

I could check out a branch:

```
fusesoc library add --sync-version feature/fusesoc-dev open-logic-fd https://github.com/open-logic/open-logic
```

And as well an older version.

```
fusesoc library add --sync-version 2.3.0 open-logic-23 https://github.com/open-logic/open-logic
```

Hence I suggest to close the issue.

rbrglez, 2025-01-06:
yes, this is exactly what I wanted :)

Closing the issue.

closed 2025-01-06 by rbrglez