olofk/fusesoc #717: Source file paths in .core files are not according to specification (not relative to the .core file)

obruendl, 2024-12-27:
According to the documentation, source file paths in .core files shall be relative to the .core file:

[image]

However, it does not seem to be implemented this way. See below.

## How to reproduce

Add the fusesoc standard core library and build the open-logic reference design:

```
fusesoc library add fusesoc-cores https://github.com/fusesoc/fusesoc-cores
fusesoc run --tool vivado --target zybo_z7 open-logic:tutorials:vivado_tutorial:2.3.1
```

You get an error although the file reported is there and the path relative to the .core file is correct:

```
ERROR: Setup failed : Cannot find vhdl/olo_base_cc_n2xn.vhd in fusesoc_libraries/fusesoc-cores/open-logic or /home/oli/.cache/fusesoc/open-logic_open-logic_base_2.3.1
```

obruendl, 2024-12-28:
I actually can make sense from the way it is implemented. One may want to write a core file for a 3rd party repo withour fusesoc support - in this case the paths can't be relative to the core file (because it does not exist within the repo).

I see two options to resolve the situation:
a) Fix the documentation and clearly state that for any core files with a provider entry, the paths are relative to the repo root.
b) Fix the implementation - search for the core file in the repo downloaded from the provider and interpret paths relative to it. Interpret paths relative to the repo root if no core file is found (I'd suggest to clearly document this as well).

Option a) clearly is less effort.

Please let me know which resolution you target, so I can adjust the core files in Open Logic accordingly.

obruendl, 2024-12-28:
The behavior is understood - Only point remaining is **updating documentation regarding file paths in .core files with provider keys** which is option (a) from above comment.

olofk, 2024-12-28:
Good point. I'll update the documentation.

One thing that might be possible is to add a parameter to the provider that just uses a subtree of the repo. We do that for the opencores provider, but not for any of the others.

olofk, 2024-12-28:
I added this now https://github.com/olofk/fusesoc/commit/48e949c9191212c322db04af3a90011855d0ac4f

closed 2024-12-28 by obruendl