olofk/fusesoc #718: Editable mode installation failing due to missing setup.py

obruendl, 2024-12-29:
I am planning to do my first contribution - I would like to add the option to specify a specific ref to use for "fusesoc library add" so users can select which release (tag) they want to add.

I followed the instructions in the [Developer's Guide](https://fusesoc.readthedocs.io/en/latest/dev/devsetup.html#setup-development-environment) to set up the development environment but installation in editable mode is resulting in an error due to the missing setup.py.

```shell
oli@Oli-Legion5:~/work/olo/fusesoc$ pip3 install --user -e .
ERROR: File "setup.py" not found. Directory cannot be installed in editable mode: /home/oli/work/olo/fusesoc
(A "pyproject.toml" file was found, but editable mode currently requires a setup.py based build.)
```

I checked the repo and there really is not setup.py - could you let me know where I am wrong?

My ultimate goal of course is to have a setup where I can run fusesoc from sources - so I can test my changes.

obruendl, 2024-12-29:
@rbrglez I noticed you recently filed a pull-request. So probably you know how to set-up the development environment - if you do, please let me know.

The change I am trying to file is related to the Open Logic .corefiles change. In that context @olofk suggested that I provide corefiles for all versions (to let users choose when to update) but this of course spams the list-cores output. If I can get my change into fusesoc, I only need to provide exactly one release version of open logic .core files (+ the dev version) and users can choose which release to use by selecting the aproopritate tag.

rbrglez, 2024-12-29:
Yeah I had similar issues with installing fusesoc in editable mode. I found out that I just need to update pip and then I could install fusesoc in editable mode.

My current version of pip, which allows me to install fusesoc is:
```bash
$ pip --version
pip 24.3.1 from /home/rbrglez/.local/lib/python3.10/site-packages/pip (python 3.10)
```

obruendl, 2024-12-31:
Was caused by my python version being old (3.8). resolved.

closed 2024-12-31 by obruendl