olofk/fusesoc #725: Question: get all the dependencies and dependency graph

ahmaddehghani, 2025-01-22:
How can you get all the dependencies of a package without executing it, and is there a way to draw the dependency graph?
thank a lot.

olofk, 2025-02-13:
Yes and no. If you add the `dot` filter (by adding `--filter=dot` on the command-line) you will get a graphviz file of the dependency tree. You will still need to run, but you can add the `--setup` flag to avoid actually building anything. It will just stop after all the project files are created.

olofk, 2025-02-13:
I also just stumbled across a really useful project called graphviz2drawio that converts graphviz files into xml files that can be loaded into draw.io for further editing. Super handy!