olofk/fusesoc #729: Question: skip synthesis stage when generate binary

leajalil, 2025-02-07:
For the tandem load flow with vivado to update regions of the fpga, a slightly different flow must be followed than normal. Two different syntheses must be generated and then the implementation and generation of binaries must be generated based on the two checkpoints created in the two previous syntheses. For this, when executing the run command, I can load the checkpoints but fusesoc tries to generate a new synthesis and I would need that stage not to be executed and go directly to the implementation. Is there any argument that allows me to do that? I use '-pnr=none' to generate only one synthesis but I did not see an option to generate only the rest.

leajalil, 2025-02-08:
The sequence i want to execute is:

```
# Read in the synthesized static netlist
add_files $synthDir/$topModuleName/${topModuleName}_synth.dcp
link_design -mode default -part $part -top $topModuleName

# Read in this version of the update checkpoint and associated constraints.
set_property HD.RECONFIGURABLE 1 [get_cells $updateInstName]
read_checkpoint -cell tandem_pcie_user_app_i $checkpoint_dir/stage2_synth.dcp -strict

read_xdc $projDir/imports/xilinx_xdma_pcie_x0y1.xdc
read_xdc $projDir/imports/${ipName}_tandem.xdc

# Write out a post link design checkpoint for this update of the Tandem 
# with Field Update design.
write_checkpoint -force $implDir/$updateName/${topModuleName}_link_design.dcp

# Run the desired implementation steps. Additional implementation steps and
# options can be added here as desired. A checkpoint is written after each
# step for convenience.
opt_design > $implDir/$updateName/${topModuleName}_opt_design.log
```

This sequence is provided by xilinx.

olofk, 2025-02-10:
Technically, this should be reported in [Edalize](https://github.com/olofk/edalize/) instead, since that's the project that has all the EDA tool interface code.

I can tell you right away though, that what you want to do is not supported by Edalize today. We could look at how to make this possible though, but I would need to understand the use-case a bit better. Like, as a user, do you only have these two dcp files, or do you want to first do synthesis to create a dcp and then link with another. A block diagram (kind of like https://edalize.readthedocs.io/en/latest/edam/api.html#vivado-flow) would be helpful to begin with.

leajalil, 2025-08-12:
I close this issue and open one in Edalize with more context

closed 2025-08-12 by leajalil