olofk/fusesoc #730: WARNING: Replacing Core:0.0 in PATH_X found in PATH_Y

schwende, 2025-02-10:
Is there a way to not search recursively for *.core files?

I have the following two libraries in my fusesoc.conf file:
`[library.work1]
location = ../
sync-type = local
auto-sync = true
[library.work2]
location = ./
sync-type = local
auto-sync = true`

As location ../ is searched recursively all the core files from work2 are also fetched and therefore I get a warning for every file. I know this can also be solved by arranging differently the folder structure, but it would be nice to have a option to specify a location without checking all the subfolders.

olofk, 2025-02-10:
If you put a file called `FUSESOC_IGNORE` in a directory, then FuseSoC will not search that directory or its subdirectories. Not sure if that helps in this case though.

If I understand your setup correctly, you could just drop work2 since it's searched by work1 already. Or am I missing something?

schwende, 2025-02-10:
> If you put a file called `FUSESOC_IGNORE` in a directory, then FuseSoC will not search that directory or its subdirectories. Not sure if that helps in this case though.
This is good to know. But in my case it's not helping.

> If I understand your setup correctly, you could just drop work2 since it's searched by work1 already. Or am I missing something?
Yes you are rigth - I simplified the example a bit too much. In fact we use like a library of cores with is located in the fusesoc folder in the project, then some files in the project folder, in where also submodules to other subprojects with also a fusesoc folder with the same library of cores in it. But it means that we just have to adapt the project structure to have core files in a seperate folder, likte that all problems can be avoided.

closed 2025-02-10 by schwende